alvarop/usb_c_cable_tester #2: Unsure if JLCPCB got this right

voltagex, 2022-11-21:
(this would be better suited for GitHub Discussions but they're not enabled here)

[image]

I've sent them your photos back - it looks like they've dropped most of the parts off the board!

alvarop, 2022-11-21:
@voltagex This is correct. The left side is what they will use :D
It is confusing though

voltagex, 2022-11-21:
[image]

To me it seems like there's only 3 components they're going to use? Some connectors are missing?

I guess I can just continue with the order and see what arrives.

voltagex, 2022-11-21:
OK, yeah, the 2D render seems correct.
[image]

alvarop, 2022-11-22:
:D

closed 2022-11-22 by alvarop

voltagex, 2022-11-29:
This arrived today. I can get it to light up by probing with a multimeter, but not with a coin cell inserted. I do not like the battery holder at all

tedder, 2022-11-29:
I would *love* to just get this off Tindie.

alvarop, 2022-11-29:
Yeah, the battery holder isn’t great, you have to make sure you go over the negative and under the positive. 
[image]

[image]