alvarop/usb_c_cable_tester #6: Consider USB-C connector with vertical pins for better soldering

vicnaum, 2022-12-13:
Consider using one of these for USB-C (where the second row of pins sticks out down and goes into the holes through the board)
[image]

Should help with soldering to get less defects (my order of 5 boards came in with 4 of the boards not soldered well 😢 and only one  was perfectly working).

I guess it's a common pain of the surface USB-C connectors - seen people complaining about them on the internet too - because the second row is hidden on the bottom of the connector and you can't really resolder it without detaching the whole thing.

alvarop, 2022-12-15:
Do you have an LCSC part number for this?
If JLCPCB has it in their supported parts library, I'd be down to switch to it in the future!

vicnaum, 2022-12-15:
> Do you have an LCSC part number for this? If JLCPCB has it in their supported parts library, I'd be down to switch to it in the future!

Yeah, there are many on their website:

- https://www.lcsc.com/product-detail/USB-Connectors_XUNPU-TYPEC-324-BCP24_C2835314.html
- https://www.lcsc.com/product-detail/USB-Connectors_XKB-Connectivity-USB31-TYPEC-32141-2FS_C319152.html
- https://www.lcsc.com/product-detail/USB-Connectors_XUNPU-TYPEC-324DBW-ACP24A_C2840399.html
- https://www.lcsc.com/product-detail/USB-Connectors_XUNPU-TYPEC-506BW-ACP24_C2895019.html
- https://www.lcsc.com/product-detail/USB-Connectors_Shenzhen-Kinghelm-Elec-KH-TYPE-C-24P-T9-87_C709359.html
- https://www.lcsc.com/product-detail/USB-Connectors_XUNPU-TYPEC-502-ACP17_C920229.html
- https://www.lcsc.com/product-detail/USB-Connectors_XKB-Connectivity-USB31-TYPEC-33141-2FS_C480368.html
- https://www.lcsc.com/product-detail/USB-Connectors_SOFNG-MC-313D_C428461.html
- https://www.lcsc.com/product-detail/USB-Connectors_XUNPU-TYC-324_C381479.html

I'm not sure what's the exact difference between then and which one is better, but plenty to choose from.

alvarop, 2023-06-11:
Hopefully fixed in https://github.com/alvarop/usb_c_cable_tester/releases/tag/v2.1

R3Q13M, 2023-06-12:
@alvarop since my fork was partially based around potential for hand soldering I spent a lot of time looking for the USB-C connector with the most easily accessible pins. I'd just like to recommend what I found since this could be a fantastic board for novice/intermediate solderers to put together by hand. 

I settled on the [G-Switch GT-USB-7056](https://jlcpcb.com/partdetail/gswitch-GT_USB7056/C963222) and would be happy to share the footprint I made with you if you're interested in it.

closed 2023-09-03 by alvarop

Ardakilic, 2024-04-06:
Hello, I've created a new suggestion for USB-C connectors. After creating the issue/suggestion I stumbled upon this issue. So since you guys know about the stuff, I'd appreciate if you could check #20 and give feedback. Thanks!

commit dd61eb28 (2023-06-11): v2.1 Update

* Updated USB-C Connector to the [Foxconn UT11113-11602-7H](https://jlcpcb.com/partdetail/Foxconn-UT11113_116027H/C2761408)
  * Hopefully this resolves #6 and #10
* Updated USB-B connector to a USB3 one [XUNPU USB-306BWD-ARW](https://www.lcsc.com/product-detail/USB-Connectors_XUNPU-USB-306BWD-ARW_C2895032.html) (Thanks @smesgr9000!) #9
* Changed USB Micro-B connector to a hopefully more robust one and also moved closer to board edge #12