alvarop/usb_c_cable_tester #7: 3D Printed/laser cut case

coryalder, 2022-12-15:
Hello!

My local hackspace recently did a group-buy of this project, and I got myself 2 of them. They works great and a will be very useful. To make sure they last me a while, I designed a 3d printed / laser cut case for it. It has a 3d printed body, and a laser cut plexy top panel so you can see the lights on the board. I just wanted to link it here in case anybody else is looking to enclose theirs.

https://github.com/coryalder/usb-c-tester-case

alvarop, 2022-12-15:
So cool! Thanks for sharing!

Going to add a link in the readme later so other folks can make their own :D

alvarop, 2022-12-16:
I've added a link in the readme. Thanks again!

closed 2022-12-16 by alvarop