alvarop/usb_c_cable_tester #8: Consider using 0.3mm vias

smesgr9000, 2023-01-04:
First of all - Thanks for sharing the PCB layout

It would be nice if the via size is increased to 0.3mm - reason I want to use PCBWay for manufacturing the PCB and cost for 5 PCBs increase from 5USD to ~50USD due to via size been smaller than 0.3mm. 

I did a quick hack to increase the via size and moved some tracks and placed an order - which is currently processed. I can share my hack if you want - but you are properly better at this than me. :sob:

alvarop, 2023-06-11:
I'll try to get this later. I just did a new spin with updated components, but didn't have the time to update all the vias. Plus the new USB-C connector probably also requires fancy boards due to the spacing :/

R3Q13M, 2023-06-12:
@alvarop thanks for looking into this. I actually forked the design for myself to address this issue as well as issue #10 but haven't gotten very far. I'd just like to suggest that good rule of thumb for via annular ring size for low-cost PCB's is via diameter plus around 12mils. So I'd recommend for a 0.3mm via a ~0.6mm diameter annular ring. 

Also, these through-hole USB-C connector holes are 0.4mm, 0.65mm annular ring, and pitched 0.8mm apart. Therefore, the minimum copper spacing is 0.15mm. For most fabs 0.15mm is about the limit for their low-cost tiers.