alvarop/usb_c_cable_tester #10: Keep cable shield and ground on separate nets

avian2, 2023-01-30:
Hi

Thanks for sharing this project. I've ordered a batch and gave the extras to friends and family.

A suggestion for the next version: keep cable shield and ground on separate nets. With v2.0 all four GND and the SHIELD LED will light up as soon as any of the ground pins and/or cable shield are connected. This is misleading, because the SHIELD will light even for cables that do not have a shield braid (i.e. do not have an electrical connection between the connector shell - this seems to be quite common in my experience).

smesgr9000, 2023-03-05:
this is an issue of the currently USB-C jack used. There is already a suggestion to switch the USB-Jack to an through hole one, which may also solve this issue.

Use the current USB-C an measure Shield with the outer pins on both rows and it will show continuity.

avian2, 2023-03-05:
You're right that the A1, B1, A12, B12 (USB C ground pins) and shield are already routed separately on the PCB. I only checked with a multimeter and haven't looked at the design files.

I find it odd that these would be intentionally shorted to shield on the connector. I don't find any note about that on the datasheet for Molex 1054500101 (the USB C connector specified in the BOM). A quick search through the USB C spec also doesn't show this (The connector should have all ground pins shorted together, but there is no mention of shorting them to shield in the connector itself). There might be something else going on.

I think I should have some USB C connectors stashed somewhere. It's not this specific type, but I'll check if those have the ground pins shorted to shield.

Looking at the Gerbers, one possibly related error I found are the bare vias under the USB C connector. These are in the restricted area of the connector. They should be covered with soldermask so that exposed copper can't touch metal parts of the connector. It's possible one of these is shorting to the shield somewhere. Since A1, B1, A12 and B12 are all shorted together in the connector, only one of B1 or B12 contacting the shield would be enough to show as short to shield from all of them. Although it would be an odd coincidence if only one of the ground pins would have a short and not any others.

Image below is from the v2.0 Gerbers. Blue is top copper, yellow is top soldermask (inverse).

[image]

[image]

smesgr9000, 2023-03-05:
no trust me - I have thorn it apart and there is a bracket which is wrapped around and have all 4 GND pins connected AND also the shield. I even tried to remove this bracket and put it back together but this would not work out.

Additionally I would say for hand soldering the connector would not be my first choice anyway.

avian2, 2023-03-05:
Odd.

I quickly buzzed out an old PCB of mine I just dug out of a drawer. It uses Wurth 632723300011. That's a USB C connector that has B1 through B12 on THT pins instead of SMT pads. It's similar to one discussed in #6.

It doesn't have the shield shorted to ground.

I can't check if A1, B1, A12 and B12 are shorted together on connector, because they are shorted on my PCB.

R3Q13M, 2023-05-09:
The Molex connector has a set of extra pins that get bridged to pins B1 and B12 because of the extra wide 'B1' and 'B12' pads in the footprint. I suspect these extra pins are connected to the connector shield. Can you modify the footprint to un-bridge these pins?

[image]

[image]

alvarop, 2023-06-11:
[image]

Confirmed, it has separate pins, but they're internally shorted. I tried them out on a disconnected part and they're still internally connected :(

R3Q13M, 2023-06-11:
Oh that's really unfortunate. Thanks for checking. Do you plan to address this issue with a new part?

alvarop, 2023-06-11:
Yeah, going to try another part for now

alvarop, 2023-09-03:
Fixed in v2.1 and later :D

closed 2023-09-03 by alvarop