alvarop/usb_c_cable_tester #13: Interpreting the results?

grymoire, 2023-06-02:
If a signal is missing, I'm not sure I can tell that there is a problem. It would be nice to have a chart that shows/lists the LED's that are suppose to be on for certain cable types.

grymoire, 2023-10-19:
A similar product has a manual, and includes a helpful guide.
[C2C-caberQU PDF](https://github.com/petl/USB-C-cable-tester-C2C-caberQU/blob/main/documentation/v3/caberQU_manual_v3.2_compressed.pdf)

Here's what page 7 shows

**USB Power Delivery**
- Option A: VBUS | GND | CC1 | Shield
- Option B: VBUS | GND | CC2 | Shield

**USB 2.0/1.1**
- Option A: VBUS | GND | D+ | D- | Shield

**USB 3.0/3.1/3.2/4**
- Single lane (3.0/3.1)
-- Option A: VBUS | GND | TX1+ | RX1+ | TX1- | RX1- | Shield
-- Option B: VBUS | GND | TX2+ | RX2+ | TX2- | RX2- | Shield
- Dual lane (3.2/4)
-- Option A: VBUS | GND | TX1+ | RX1+ | TX1- | RX1- | TX2+ | RX2+ | TX2- | RX2- | Shield

**Debug Accessory Mode**
- Option A: VBUS | GND | D+ | D-| CC1 | SBU1 | TX1+ | RX1+ | TX1- | RX1- | Shield
- Option B: VBUS | GND | D+ | D-| CC2 | SBU2 | TX2+ | RX2+ | TX2- | RX2- | Shield

**Alternate Mode**
- Option A: VBUS | GND | TX1+ | RX1+ | TX1- | RX1- | Shield
- Option B: VBUS | GND | TX2+ | RX2+ | TX2- | RX2- | Shield

**Audio Adapter Accessory Mode**
- Option A: GND | CC1 | CC2 | SBU1 | SBU2 | D+ | D- | Shield
VBUS is optional for concurrent charging of a device.

NSExceptional, 2024-01-11:
It's a shame the author hasn't responded. Is that chart enough for a lowly coder to use to organize his USB cables in your opinion, @grymoire?

ejl371, 2024-01-26:
Wikipedia tells you what pair does and what modes it is used in.  https://en.wikipedia.org/wiki/USB-C.  The author is amazingly responsive when I reached out to ask about another issue.  Which coincidentally someone else opened the same day (issue 18).

anyconhhv, 2024-01-29:
I gathered all of my cables and tested them. I then attached a label to each with the type of USB. And then if a cable doesn't work, I try a cable with more connections.  There is an amazing number of variations in type and price, especially in USB-C cables. I have photos of a couble that might help.

This first one was an expensive cable -  All lines are on exzcept for D+/D- on one side, but if you flip the USB3 cable over, the D+/D- on the other side lights up.
[image]

Now here's cable #2. I am not sure how this cable was classified. Note that how I plug it in, there are three different combination of light patterns.
[image]

Now flipping over the USB-C, note how TX2/RX2 become TX1/RX1
[image]

And now plugging into the green USB2 port instead of the blue USB3 port, we just get VBus/Gnd and D+/D-. Note we have 2 sets of Gnd/VBUS/
[image]

Here's a USB-C to USB-C cable I use for fast charging. Note it has two sets of VBus/Gnd and D+/D-, but there is also a CC2 light. If I flip the cable over, CC2 on the left changes to CC1 on the right.

[image]

And finally - a plain, ordinary  USB2 data cables -  with one set of VBus, ground and D+/D-. 
[image]

grymoire, 2024-01-29:
Oops. I submitted the above info on my other account. I'm also [anyconhhv](https://github.com/anyconhhv)