alvarop/usb_c_cable_tester #16: Manufacturing defects on the new USB-C receptacle

FHeilmann, 2023-07-19:
Hi, thank you for sharing this board!

Unfortunately JLCPCB seems to have issues assembling the new USB-C receptacle. I've ordered 5 assembled PCBs and while the USB2.0 and even the 3.0 A-to-C works ok on 3 or 4 of them, pretty much all of them have issues with the 3.0 C-to-C cable testing.

With a known good USB-3 C-to-C cable, each board shows a different combination of LEDs, on some, even the basic pins such as D+ and D- stay dark, which leads me to believe that the assembly of the Type-C receptacles is faulty.

alvarop, 2023-07-29:
Hey! Yeah, it was a bummer. I think it would work better with a 0.8mm PCB.
I still need to do an order with that one and see if it helps.
I was able to "fix" many of them manually with a soldering iron but it's a pain...

The previous connector that was SMT only was great, but it shorts all the ground pins together with the shield :(

alvarop, 2023-09-03:
I changed to use 0.8mm PCB's (instead of 1.6mm) and had MUCH better results. 95% yield :D
[image]

1/20 boards had some solder bridges on the USB-C connector, but it was easy to fix.
[image]

Not all of the boards had the soldermask missing between the pins though. Might have to slightly change the footprint to improve the chances of the mask being there.

[image]

The through hole pins had much better soldering on them:
[image]

closed 2023-09-03 by alvarop