alvarop/usb_c_cable_tester #18: Cannot order from JLCPCB because FOXCONN UT11113-11602-7H is unavailable

MacLemon, 2024-01-24:
Thanks for the excellent instructions on how to order a populated PCB through JLCPCB. They help people like me who aren't familiar with PCB making and ordering.

Connectors `J1`, `J2` are a `FOXCONN UT11113-11602-7H` which hasn't been in stock at JLCPCB for about two months now. As a consequence I'm unable to order a PCB since I lack the skills to pick a suitable replacement receptacle.

Would it be possible to update the BOM with an alternative Type-C receptacle that is available or use a different PCB manufacturer instead?

alvarop, 2024-01-24:
Hi Marc,
This is the unfortunate side effect of using whatever was available at the time. I'll have to redesign the board to use a more commonly used connector. I may be able to do it this weekend, but I'm not sure.

I'll make sure to update here and add a new release when that happens.

lnrdmx, 2024-01-26:
[image]
same, appreciate your work alvarop!

alvarop, 2024-02-11:
Hi again.

Taking a look at the JLC USB-C connector inventory and it's not great. There's only one option that even has more than 1000 in stock. I'm afraid this will become never ending if I have to keep re-designing whenever a connector goes out of stock.

The other option is to design it with commonly available connectors, which means the price will be much higher and manufacturing process not as simple, since they won't be in the JLC parts library.

At this point, if you really need one, you can get something on amazon for cheaper: https://www.amazon.com/Treedix-Cable-Checker-Charging-Type-C/dp/B0BW8ZC7YD

ejl371, 2024-02-11:
Thank you for looking into it.

ejl371, 2024-02-19:
FYI, The USB C connector is back in stock.

alvarop, 2024-02-20:
Thanks for the update!

alvarop, 2024-08-17:
Closing this one. If they're available, that's great, but if not, you can build v2.4 with more commonly available parts here: https://github.com/alvarop/usb_c_cable_tester/releases/tag/v2.4

closed 2024-08-17 by alvarop