alvarop/usb_c_cable_tester #20: Suggestion: Change USB-C connector to a widely available and cheaper one

Ardakilic, 2024-04-06:
Hello,

First of all, thank you for open-sourcing the project.

Today I have built the tester myself, and realized the usb-c connector that I've purchased does not fit properly to the case.

This is because the pinout is somehow different.

After confirming the pins match, I have bought this: $1.91 / 10 pcs:

https://www.aliexpress.com/item/32879436811.html 

[image]

[image]

[image]

However, today I realized the ground shielding does not fit. 

I used a wrench and my soldering techniques. It fits and works.

[image]

[image]

To change to the proper connector, I searched, and in Aliexpress, there are only 2 sellers, which I believe sells the item and fits to the footprint. I thought these should be widely available.

The closest one, which I could find available is this, and it costs $1.4 per part, https://www.aliexpress.com/item/1005005376413109.html

And it's almost 10 times more expensive than what I've bought, even more expensive than LCSC.

There are various connectors that are 24 pin and pinouts match 1:1, and available at LCSC, where the footprint matches, widely available in other vendors, and they are way cheaper:

For example:

* $0.52 per connector, and the most available part on the category: https://www.lcsc.com/product-detail/USB-Connectors_XUNPU-TYPEC-324-BCP24_C2835314.html
* $0.25 per part - https://www.lcsc.com/product-detail/USB-Connectors_Lian-Xin-Technology-XUBF-0336-B07_C7527707.html
* $0.46 per piece - https://www.lcsc.com/product-detail/USB-Connectors_SHOU-HAN-TYPE-C-24P-QCHT_C456013.html
* 0.45 per piece - https://www.lcsc.com/product-detail/USB-Connectors_Hanbo-Electronic-MC-263-H20_C6332283.html
* $0.36 per piece https://www.lcsc.com/product-detail/USB-Connectors_HCTL-HC005-1A1H1A103-0HR_C2894904.html
* Discontinued, but you get the idea: https://www.lcsc.com/product-detail/USB-Connectors_XUNPU-TYC-324_C381479.html

The greater issue is that the lack of availability of the part, apart from the price, as I mentioned.

I reside in Turkey, and my best bet is Aliexpress. Afther that there is Farnell, which they don't stock much, and ozdisan.com , and they both lack the connector.

And in Aliexpress, I found only two sellers, as I mentioned.

The pinout is same for both 24 pins, SMD 12 pins, and the through hole 12 pins, so I see no problem on switching the footprint to this widely available footprint model.

I'd appreciate if this is considered.

Thanks in advance!

exetico, 2024-05-04:
Hi, 

I just stumbled upon the project, and noticed your issue, as I noticed the pretty high price of the C2761408, currently used in the design (8,43 USD for 10 Qty).

So I'll just bump this, so I can follow the process. It's the most expensive part of the BOM, so I guess there's many options to bring down the cost a bit, by replacing the connector with a cheaper one, as you're also suggestion. I'm not going to risk ordering it with another connector, without having a input from the original author, though. 😄 

Thanks for bringing this to the table.

Ardakilic, 2024-05-08:
@alvarop Thanks for the reply. 

The main issue I find with the current footprint is that it's not a rather generic one, so you're stuck with less brands (both in JLC/LCSC and other vendors). All of the links I've shared (and even more on LCSC's inventory) share the same footprint. That's why I suggested in the first place. 

Of course, the price is a bonus, however, my main concern is the availability, and as you mentioned, current stock does not represent the future stock, that's why I believe a more generic footprint brings more alternatives and availability to the table.

alvarop, 2024-08-17:
I've created a hand-solderable one with commonly available parts here: https://github.com/alvarop/usb_c_cable_tester/releases/tag/v2.4

It's more expensive, but it won't require chasing component stocks as much.

closed 2024-08-17 by alvarop

Ardakilic, 2024-08-17:
Heya @alvarop , awesome to hear there's a hand solderable version now. Thank you for the heads-up, and thanks again for the work you've put into this!